CMU-Robotics-Club/RoboBuggy #123: Arduino mega does not break out enough input capture pins

ihartwig, 2015-02-21:
We can only get to ICP4 and ICP5, but we need to read 3 RC signals. I think we should do this with either standard interrupt pins (what we're doing now) or a pin change interrupt mask.

ramensandwich, 2015-02-21:
What are the three signals which we'll be reading?

ihartwig, 2015-02-21:
I noticed this while working on the hardware usage map (https://github.com/CMU-Robotics-Club/RoboBuggy/wiki/Hardware-Usage-Mega). That might help make this more clear. We are sending out 1 servo pulse (steering servo) and reading 3 of them (steering, brake, auton state) from the RC receiver. Each 16-bit timer can send 2 or 3 in hardware depending on the number of output compares it has, but it can only read 1 signal on the 1 input capture pin. Unfortunately, the arduino does not connect ICP1 or ICP3 to a pin header.

We could add a header and solder directly to the pin. I think we can make this robust enough, but I'm not sure it is worth the effort.

TrevorDecker, 2015-04-05:
has this problem been solved?

ihartwig, 2015-04-06:
It has been resolved-ish in that we wrote new rc receiver code based on
external interrupts rather than input capture. Based on watching the
interrupts on a scope and actual operation, this should work fine. We're
waiting on a full serial reading implementation before closing this.

TrevorDecker, 2015-09-14:
fixed (we think)

closed 2015-09-14 by TrevorDecker