CMU-Robotics-Club/RoboBuggy #142: Record video from web cams

TrevorDecker, 2015-03-29:


TrevorDecker, 2015-09-14:
The entire gui should be redone

kbrennan234, 2015-09-16:
I started work on a new gui using JavaFx, but it still needs some work
flushing out the internals. Most of the layouts are similar to what we
currently have, but JavaFx offers more support in developing more
professional looking gui's and 3D drawing tools. Also, it switches to using
css tables for style like in web design.

If you are interested, I can send the code, and you can choose what you
want to take.

TrevorDecker, 2015-09-16:
That sounds awesome!! Can you make a new branch and push it to the repo? 

Also hope your summer/postgrad life is going well.

TrevorDecker, 2016-02-13:
This has been implemented in pure java

closed 2016-02-13 by TrevorDecker