CMU-Robotics-Club/RoboBuggy #147: fix surface mount for old buggy

TrevorDecker, 2015-04-05:


TrevorDecker, 2015-08-29:
no longer needed as old buggy is being deprecated

closed 2015-08-29 by TrevorDecker

ramensandwich, 2015-08-29:
Are we not keeping old buggy as a testbed?

TrevorDecker, 2015-08-29:
The main focus this semester will be on transistor. If we come up with a good reason to keep nixi up to date then we can readjust our priorities. But considering the size of our current todo list and the possibly large over head of maintaining a second buggy I think we should focus all efforts towards transistor.

VasuAgrawal, 2015-08-29:
"nixi"?

TrevorDecker, 2015-08-30:
Nixi was my name for singularity, nixi tubes were used before transistors.  We used Nixi before transistor.

ihartwig, 2015-08-31:
Nixi is probably more aptly a predecessor to LEDs than transistors btw. Vacuums tubes aren't the same as nixi tubes.

TrevorDecker, 2015-08-31:
Ian is correct but "vacuum" or "vacuum tube" doesn't sound as good as a buggy name.  So I am using a small artist license. Keep in mind nixi tubes did come before transistors while the singularity won't happen for many years