CMU-Robotics-Club/RoboBuggy #150: add potentiometer to new buggy steering column

TrevorDecker, 2015-04-05:


ihartwig, 2015-06-03:
Suggested implementation in software: low level should run a "task" that acts as a buggy monitor. This should shut the buggy down (drop brake) if battery is low, pressure is insufficient for braking, lost signal, etc.

TrevorDecker, 2015-09-26:
@ihartwig what does your comment have to do with a potentiometer being added to the buggy?  Are you suggesting that we have a check to see if the potentiometer disagrees with the command by angle by more then a threshold? 

Right now the servo has been added, mechanical wants to make the mount more secure.  We still need to implement reading the pot in software.

TrevorDecker, 2015-10-11:
This has been done and we are currently reading values from the potentiometer.

closed 2015-10-11 by TrevorDecker