CMU-Robotics-Club/RoboBuggy #156: Fix serial reading

ihartwig, 2015-06-03:
Right now, serial reading drops packets. This is because we spin in a tight loop around waiting for available bytes, but processing each byte takes longer than the time between bytes coming in. This is because the AVR hardware has no serial buffer. Arduino implements a small circular buffer filled by interrupts. We should do something similar.

This is blocked by the part where UART is hardcoded.

ihartwig, 2015-06-03:
@ramensandwich

ihartwig, 2015-07-01:
avrlib already has a lot of #defines and macros that allow us to access the registers and make interrupts. However, this is not very object oriented. The easiest way I can think of to make this pretty is to pre-allocate buffers and interrupt routines that are enabled based on the hardware selected. Then, an object oriented layer can sit on top that just holds pointers to the static buffers and functions.

Since this is a lot of bit twiddling that depends on specific chips, we can reuse someone else's hard work: https://github.com/andygock/avr-uart

TrevorDecker, 2015-09-26:
was this fixed with #163  or is it still a problem?

TrevorDecker, 2015-10-11:
Fixed with new serial reading lib

closed 2015-10-11 by TrevorDecker