CMU-Robotics-Club/RoboBuggy #162: Slim down git download history

TrevorDecker, 2015-08-22:
We have bunch of commits and merges that slow down the download and install time of this repo

TrevorDecker, 2015-09-14:
@msebek  can you give an update on this, I remember you mentioning something about removing opencv from the code

msebek, 2015-09-29:
After rather extensive investigation, it looks like this is going to be Hard. We could remove the whole source code of libraries that were added to the repo, but this involves rewriting history (thus breaking anyone who is branched from master/who has pulled ever). 

In addition, this is not foolproof; history rewriting does not have a concept of removing files across renames, so removing whole libraries is highly laborious and error-prone. The maximum gains I was able to achieve this this approach was making the repository 200mb (from the original 400mb). 

The best way (especially time-efficient) for us to get a 10x (likely a 100x) improvement would be to, at 'some point in time', copy the files to a new repo, and lose all history. We could keep the old (large) repo around to be able to hunt down regressions, while committing new code and bugfixes to the new repo. The transition period would have to be chosen carefully, since it would require all contributors to synchronize and cherrypick their work on top of the new repo. 

Or we just keep the current repo. Which is probably the solution we will keep until (at very earliest) Winter Break, more likely after RaceDay 2016, and very possibly forever.

closed 2015-09-29 by msebek