CMU-Robotics-Club/RoboBuggy #186: E brake deployed when buggy moves

TrevorDecker, 2015-09-20:
The current version of master causes for the E brake to deploy while the buggy is moving.  Observed during walk around on 9/20 at the end of hill 2.  

Re reverted to the code run during raceday 2015 and the problem went away.

ramensandwich, 2015-09-21:
Do we know what version of code was running on the buggy?

TrevorDecker, 2015-09-21:
I think it was what is currently running on master @ihartwig  would know for sure

ramensandwich, 2015-09-21:
Too bad there was no versioning system of some sort to keep track of such information.. :)

TrevorDecker, 2015-09-21:
That would be nice, and I hear that we will have it soon.  But in this case it would not help since we flashed a different version of the firmware on to the buggy for our roll around.  If we really want to track this type of thing we should have the high level system log the low level systems version number every time data is recorded.  NOTE this is totally doable once we get low level version number reporting.

ihartwig, 2015-09-25:
Isolated this to versions of master after commit ba823582aa54b172d71c9e5891d312206791572c which merged 2bf068694af1469895a957a619e1be7f9d1aa339 into master. This commit reverted a commit which fixed issue #134. A revert revert is pending in pr #192.

Root cause: poor git practice. Please be careful what we merge. This is (1) a ramification from gitpocalypse and (2) the result of an un-reviewable 300,000 line pr. Please stop.

TrevorDecker, 2015-09-26:
This was not an issue at rolls 9/26/15

closed 2015-09-26 by TrevorDecker