CMU-Robotics-Club/RoboBuggy #202: Send compilation information out from arduino

ihartwig, 2015-10-02:
PR #161 added compilation information in a generated fingerprint.h file. We should make it send the compilation hash number all the time and printf all the information at startup.

ramensandwich, 2015-10-02:
Does this mean that the Surface will have to be running before the Arduino runs? Or will the information be sent as soon as a connection over UART is made?

ihartwig, 2015-10-04:
You'll have to be listening when the arduino starts to get the full human-readable dump, but the surface can get the basic information (hash/clean) by the rbsm sent all the time. This will let it check against a known good builds list. It's similar to the rbsm for device id.

ramensandwich, 2015-10-05:
How are we going to manage the list of good builds? Are we going to create a new file somewhere of roll-ready builds?  Going a little further with that, if no one has any objections, on the high-level side I think we should put up a warning (not error) on the GUI or somewhere visible when we're running a non-approved build

TrevorDecker, 2015-10-05:
I agree with this, Sean do you feel comfortable doing the high level and
low level portion of this implementation.

ramensandwich, 2015-10-05:
I'll try to have a build ready this week.  I'll need to bug someone to test it for me though since I don't have easy access to the buggy :)

ramensandwich, 2015-11-22:
Resolved with PR #228

closed 2015-11-22 by ramensandwich