CMU-Robotics-Club/RoboBuggy #203: Configure the GUI to not crash whenever we open it with buggy not connected

VivaanBahl, 2015-10-02:
This works on Vasu's computer but not on the surface. Would be nice to figure out what's going on

VivaanBahl, 2015-10-09:
```
Starting GUI
Populated!
Starting Logging
Initializing IMU Serial Connection
Stable Library
=========================================
Native lib Version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
Java lib Version   = RXTX-2.1-7
WARNING:  RXTX Version mismatch
	Jar version = RXTX-2.1-7
	native lib Version = RXTX-2.2-20081207 Cloudhopper Build rxtx.cloudhopper.net
Unable to connect to necessary device on COM6
gnu.io.NoSuchPortException
Trying to close the window!
	at gnu.io.CommPortIdentifier.getPortIdentifier(CommPortIdentifier.java:218)
	at com.roboclub.robobuggy.main.mainFile.connect(mainFile.java:67)
	at com.roboclub.robobuggy.main.mainFile.bringup_sim(mainFile.java:123)
	at com.roboclub.robobuggy.main.mainFile.main(mainFile.java:52)
```

TrevorDecker, 2015-11-15:
This has been fixed in logging play back

closed 2015-11-15 by TrevorDecker