CMU-Robotics-Club/RoboBuggy #211: Encoder shows ticks when wheel is not moving

TrevorDecker, 2015-10-10:
For certain wheel positions the encoder is registering ticks when the wheel is not moving.  It is believed that this is happening because the encoder is on the edge of a wheel spoke causing for the ir beam to osculate between high and low. 

Possible fixes: 
1. detect when the buggy is not moving

2. Start the buggy so that it is not at one of the edge cases 

3. Add a second encoder to do data validation
 
4. Add a comparator and have a larger dead band 

5. Ignore it and let the state estimator work things out.

spencerearly, 2015-10-11:
I think a comparator with hysteresis might be the best solution.  We could
have something like this:
http://electronicdesign.com/site-files/electronicdesign.com/files/archive/electronicdesign.com/content/content/63189/63189-fig3.jpg

The other possible fixes seem either unnecessarily complex (adding 2nd
encoder) or don't really solve the problem (avoid starting the buggy at an
edge case).  I don't understand the state estimator, so that might work.

Could someone show me a schematic of the current encoder setup?

TrevorDecker, 2015-10-11:
@zsd-planekid If I remember correctly you were the creator of this sensor.  Do you have documentation on it available online?

TrevorDecker, 2015-11-15:
This has been fixed with a schmitt trigger

closed 2015-11-15 by TrevorDecker