CMU-Robotics-Club/RoboBuggy #224: Phantom GPS

VasuAgrawal, 2015-10-27:
Upon disconnecting the GPS, we're still getting constant reads.

Trevor and I were able to fix this by changing how we read to using event based code. We're going to need to rewrite the serial parsing, which is just more copying code off of google.

TrevorDecker, 2015-10-28:
@VasuAgrawal This needs to be fixed.  Won't fix is for issues that we are not going to fix in the near future.

VasuAgrawal, 2015-11-06:
Looks like rxtx handles the reconnection on the port for us. I'm just going to make it work for the same port for now, and will try to get it to work for general ports later ... but I'm not going to do the latter until we attach header bits to every stream.

TrevorDecker, 2016-02-07:
This does not seem to be an issue I also moved us to a new serial reading library

closed 2016-02-07 by TrevorDecker