CMU-Robotics-Club/RoboBuggy #225: GPS isn't logging.

VasuAgrawal, 2015-10-27:
For some reason, on both the current version of master and on my dev branch, it seems that the GPS isn't logging. We've stumbled across this before, and it's probably just because we're in the club and don't have a lock.

That said, if this issue persists then we know it's an issue with GPS node since both master and my branch exhibit the same behavior.

VivaanBahl, 2015-10-27:
I've noticed this too, the cause seems to be that we start logging before GPS locks

msebek, 2015-10-28:
Note that if the GPS is not flashing the blue light at about ~1hz (read: the GPS does not have a lock), then there is no lock, and there will be no messages. If the GPS does have a lock (as visually confirmed by the flashing blue light)

VivaanBahl, 2015-10-28:
Right, I've seen quite a few logs unfortunately go without gps data because we power the buggy on and go around the course right away without getting lock.

TrevorDecker, 2015-10-28:
The system should be able to handle us starting logging before we get a gps
lock.  The gps is outputting data before a lock is established, all the
values are just blank.   The system should also be able to handle the gps
being unplugged and replugged in while logging is running.  Remember the
idea is to make it so that we can't mess it up at 5:00am.

VasuAgrawal, 2015-11-05:
@msebek we're still actually receiving messages from the GPS. There's a quality bit that we're simply not using. I don't think our parser is correct (fully).

TrevorDecker, 2015-11-15:
gps currently logging, reconnect is being handled by a another issue

closed 2015-11-15 by TrevorDecker