CMU-Robotics-Club/RoboBuggy #237: Reducing gps lock time

TrevorDecker, 2015-11-07:
This morning we had trouble getting a gps lock.  It took about 5 minutes while in the middle of the road. We also seem to be getting far fewer pings then expected.  

I am opening up this issue to start a discussion about ways to reduce the lock time and increase the read rate.  We may want to investigate adding a second gps unit to the buggy. 

Note this is not a slack thread as I want it to be a more targeted discussion. Feel free to talk about this in slack or in person but please add conclusions here.

VivaanBahl, 2015-11-07:
Maybe we could mount the antenna to the outside of the buggy, like on the pushbar?

VivaanBahl, 2015-11-08:
So during rolls this morning (11/8/15) we decided to test this out. We put the gps antenna to the pushbar on the outside of the shell, and we were able to obtain lock in roughly 10 seconds.
Also, during rolls yesterday (11/7/15) we had about 97 GPS data points in 13 minutes (nowhere close to the projected sample rate), but with this new solution we got 406 data points in 7 minutes, which is not that far off from the projected value of 420. Dani and mechanical are currently working on a mount that will attach to the pushbar which will now house the antenna, and we'll widen the camera cable hole in the back to support the antenna cable too.
This basically means that the GPS is now reliable (and will probably have more precise data).

closed 2015-11-08 by VivaanBahl