CMU-Robotics-Club/RoboBuggy #312: system deadlock

TrevorDecker, 2016-02-13:
If too many messages are published before the gui starts then the system seems to lock up.  It seems that the publishing thread and gui both stop operating

TrevorDecker, 2016-02-14:
This has been observed on trevor test 

Messages also appear to stop being published after about a minute of running.   

I think this has to deal with us not dropping messages (note I still need to look into how buggyros deals with having too many messages)

TrevorDecker, 2016-02-15:
This was partially solved by changing to a fixed size queue, Ideally this will be changed to a stack

TrevorDecker, 2016-02-20:
@msebek

VasuAgrawal, 2016-02-29:
@TrevorDecker, what is the fixed size queue used for? @VivaanBahl  is looking at using a fixed size queue for either buggyros or the logging interface, but if it's already implemented somewhere then we can match the implementations.

VivaanBahl, 2016-02-29:
@VasuAgrawal This is a different issue than what we were talking about in #342.

msebek, 2016-03-02:
Publishers publish into a central message queue, implemented as a blocking deque with a single thread that services it. This mailman thread dequeues a message from the central queue, and pushes it onto the queue of each subscriber. Each subscriber has an internal message queue, and a single thread that services it. This subscriber thread dequeues a message, and runs the user-specified callback. 

Trevor limited the mailman queue to be 20 (or was it 40) messages long. I'm not sure what effect this actually has on the system, but it seems to mask the problem. 

A different thing to try would be limiting the number of messages in the Subscriber, and removing the limitation on the central queue. This would be 'fairer', in that a single poorly-behaved subscriber would not be able deny service to all other subscribers. 

Additionally, someone should check that we are dropping messages that don't have any subscribers to them. If we are not, this would explain the "too many messages are published before the GUI starts up".

TrevorDecker, 2016-03-03:
so my solution was a hack, lets try and get a better long term solution

TrevorDecker, 2016-03-17:
@msebek did you ever find a good solution to this?

TrevorDecker, 2016-04-01:
this has been fixed and merged into master

closed 2016-04-01 by TrevorDecker