CMU-Robotics-Club/RoboBuggy #316: Slow ping to surface

VasuAgrawal, 2016-02-13:
In testing yesterday, the surface wasn't responding within acceptable margins for ping. Ping varied from 2ms to 3000ms. This is possibly due to a poor router. Determine if new high end hardware is necessary, and if so, find a part number.

TrevorDecker, 2016-02-15:
It was suggested by @msebek that this problem exists because we are sending too much data across the network. We should rerun this test when only the server is connected but no data is being published

VasuAgrawal, 2016-02-24:
I'm not sure that's correct. AFAIK we ran this test with the buggy program not running, so the only things on the network should have been the surface and the ping. Will confirm later.

TrevorDecker, 2016-03-03:
lets re run this test, also we have a new router coming in soonish

VasuAgrawal, 2016-03-07:
Will re run when the new router comes in. When is the mobile broadband solution expected? we can test with that as well.

TrevorDecker, 2016-03-07:
router should be in for next rolls, broadband will likely not be ready for
3+ weeks

VasuAgrawal, 2016-03-07:
Cool, I can try to get that tested some time this week if USR isn't bad.

closed 2018-04-13 by VivaanBahl