CMU-Robotics-Club/RoboBuggy #338: GUI Quitting Unexpectedly

neimandavid, 2016-02-27:
When I run the GUI on my computer, it quits unexpectedly.

The error message on the Java console is:
objc[3019]: Cannot form weak reference to instance (0x7fd89a45e910) of class _CSIRenditionBlockData. It is possible that this object was over-released, or is in the process of deallocation.

(The memory location changes between runs. There's then a bunch of Java stuff saying what looks to be generic "stuff went wrong" information.)

ramensandwich, 2016-02-27:
If it suggests that the memory location changes between runs, that implies a concurrency issue (i.e. something tried to free an object and another is trying to use it)

VasuAgrawal, 2016-03-05:
@neimandavid , does this happen every time, or is it a rare occurrence? Can you try trevor-test and see if that produces the same bug?

neimandavid, 2016-03-06:
I probably won't get around to testing this any time soon, but it was
happening every time I ran the GUI (just at varying times). I don't
remember what I was running (probably my branch and not trevor-test), but
I'll let you know when I get around to testing it.

closed 2018-04-13 by VivaanBahl