CMU-Robotics-Club/RoboBuggy #339: video files  are corrupted and we need a way to uncorrupt them

TrevorDecker, 2016-02-28:
Many of our video files will become corrupted when they are not closed properly.  For mp4's the data about the file is at the end.  the moov section. 

we need a way to recover these videos.  one solution would be to use avi which saves this data in the beginning.

VasuAgrawal, 2016-02-29:
Is using a format like avi possible, if we're recording an unknown amount of data?

TrevorDecker, 2016-03-03:
I think its possible, however mp4's seem to be more space efficient then avi's

msebek, 2016-04-01:
Proper support for shutdown/close may also solve this?

VivaanBahl, 2016-04-01:
Yeah because this issue happens when the JVM crashes. If you look at the accompanying log files with these corrupted video files you'll see there's no footer object in the json

closed 2018-04-13 by VivaanBahl