CMU-Robotics-Club/RoboBuggy #342: Heap overflow @ 720p

VasuAgrawal, 2016-02-29:
Last night, I was able to get the JVM to run out of heap space when recording data at 720p instead of whatever the default resolution is. This error does not occur on my laptop and I believe it to be a hardware limitation, but there's a possibility it's also a software thing with us not handling the images as efficiently as possible.

This was before I had pressed the start button; are we storing images to be logged somewhere? If we are, then they need to be discarded because otherwise the heap will overflow.

VivaanBahl, 2016-02-29:
The way we wrote the system, things should not be getting stored unless we are logging, but given that the GPS seems to be getting stored somewhere in the background too (issue #341), you may have hit a bug in the logging library. I'll check this out later this week.

VasuAgrawal, 2016-02-29:
Sounds good. You should hopefully be able to find it by looking at all of the places in the code where we're bulk subscribing to things; that's *probably* where the bug is.

VivaanBahl, 2016-02-29:
Yeah I think it may be something like that, because we bulk subscribe to the channels when we instantiate the Logging node, not activate it. This may be where the problem lies, because the subscribers are probably adding to the message queue before the button is pressed

VivaanBahl, 2016-02-29:
Yup, this is the case. We call the method `setupSubscriberList()` from the constructor. Each subscriber when triggered, adds its message to the queue. I'll try to push out a fix soon

VasuAgrawal, 2016-02-29:
Another simple fix would be to use one of the java objects that has a queue size limit so that we don't buffer more than we can handle.

VivaanBahl, 2016-02-29:
Well the thing is that while that would solve the data overflow issue, it still wouldn't be the expected behavior of the system. We want the behavior of the system to be that it only writes messages to the log file that came in between pressing the start button and pressing the stop button, and right now the system has the potential to buffer up a lot of messages that will get written even though we haven't yet pressed the start button.

VasuAgrawal, 2016-02-29:
Good point. I'd argue that we should put that in there anyway as a safeguard, as it's arguably more important to have a running system than to have logs and a system that has the potential to break itself. This is so that if we hit this bug again somewhere else it'll be less of an issue.

VivaanBahl, 2016-02-29:
sounds good, ill try to implement both later this week

VivaanBahl, 2016-03-03:
This issue may be fixed by 51cbf64f, will need to test to make sure

TrevorDecker, 2016-03-03:
lets do a call/meeting about this over the weekend, I would prefer to limit the number of random queues we have in the system.  ^^ the queue I added in the deadlock issue was meant to be a fix just for that day while matt fixed the underlying issue

TrevorDecker, 2016-03-21:
what is going on with this?

VasuAgrawal, 2016-03-26:
At some point @VivaanBahl  said he was going to make it so that messages aren't even put into a queue unless logging is active. I think this is now the case by initializing the subscriber list once the start button is pressed, but someone should correct me.

Qualitatively, I haven't seen this bug working on the most recent version of master.