CMU-Robotics-Club/RoboBuggy #343: Create testing checklist

VivaanBahl, 2016-03-03:
Normally when software goes out to test we forget to do something that sets us back by about an hour. Previous instances have included:
- Forgetting to turn on the radio before internal power (snaps steering linkage)
- Needing to put the brake override in rather than needing to pump and forgetting the pump
- Clearing the JVM's lock on serial ports after a faulty disconnect
- using the wrong COM ports on the system
It would be nice if we had a small checklist that we run through before we take the buggy out to test, so we can skip debugging issues we shouldn't and get to actual testing

TrevorDecker, 2016-03-03:
I agree that a check list would be nice, and useful to get people trained on how to operate the buggy.  But I really think that we should fix the root of the problems. 

https://github.com/CMU-Robotics-Club/RoboBuggy/wiki/Testing-Procedure

ramensandwich, 2016-03-03:
I like the idea of a checklist or a guide book!  We can create a systematic approach for prepping, running, and testing the buggy much like flight controllers do

TrevorDecker, 2016-03-07:
If we do this I would like to have records kept about which steps fail.  For instance we should log on when a specific sensor failed.