CMU-Robotics-Club/RoboBuggy #348: JFree Chart throws errors

TrevorDecker, 2016-03-10:
We are moving to a proprietary graphing system to fix this problem.  The issue is that JFreeChart is not thread safe.  Luckily this is not a critical bug, it just fails to update the graph on some updates, but will get refreshed on the next update

msebek, 2016-03-10:
Note that this could be solved by a JFreeChartGlobalLock that you acquire before draw and release after. 

(assuming a few things, such as everyone who draws on graphs is equally important)

TrevorDecker, 2016-04-03:
It is also believed that JFreeCharts are being greedy and causing memory leaks.  I have John working on implementing an in-house  plotting solution

VasuAgrawal, 2016-04-03:
Is this still necessary? I feel like we're applying hotfixes to something that will be osboleted quickly. It would be faster and more efficient toward the end goal to have John work on getting graphs in the web gui, and remove them entirely from the gui on the surface.

TrevorDecker, 2016-04-04:
I want to have graphs in java since the bring up time for getting a graph from java will be faster then for the web in term of developer time.  Lets say we want temperature vs speed or something weird, I could imagine this being a 1 liner in java.  It would be much more then that with our current infrastructure for the web.