CMU-Robotics-Club/RoboBuggy #355: Remove all prints that are outside of RoboBuggyLogic

TrevorDecker, 2016-03-16:
Ideally this will be implemented by a style check which will not pass any code that has a print outside of buggyLogic.  

This is low priority but will help with debugging

TrevorDecker, 2016-03-21:
this is addressed by #374, currently waiting on review to be pulled into master

VasuAgrawal, 2016-03-26:
there's currently a few other prints that are happening due to the various libraries we are using. Those should be caught and eliminated. Bing maps is a repeat offender.

TrevorDecker, 2016-03-26:
This has been merged into master 

The core functionality has been implemented I am going to open a new issue for bing maps. 

Really the way I implemented this is a little hacky but it does work.  The intent of the issue was to stop debug statements from being added to master

closed 2016-03-26 by TrevorDecker