CMU-Robotics-Club/RoboBuggy #386: Use a more powerful microcontroller for low level

ihartwig, 2016-03-27:
High-resolution steering and encoders are pushing the limits of the Arduino Mega. Running the steering motor at > 25% power (about 400 deg/s) consumes 50% of the cpu time with quadrature encoder readings. This doesn't include distance encoder interrupts, RC decoding interrupts, or main loop processing. Our main loop targets 100Hz but yields to interrupts first, which can cause it to be starved.

If we want to move to a new hardware platform it should support
* 5V IO
* PWM generation
* Pulse time reading on multiple channels
* Quadrature (and single) encoders
* 32-bit processing
* Loading programs with an internal programmer
* Out of the box support for an RTOS

ihartwig, 2016-03-27:
The best option for this off the top of my head is [Cypress's PSoC 5 Dev Kit](http://www.cypress.com/documentation/development-kitsboards/cy8ckit-059-psoc-5lp-prototyping-kit-onboard-programmer-and). It is 32-bit ARM with programmable hardware blocks to meet the needs above and supports 5V IO. The dev board is easy to breadboard/protoboard and comes with an onboard programmer/JTAG adapter. The IDE is only available for Windows ( =/ ) but is really helpful.

msebek, 2016-04-01:
did the stm32f4 discovery dev board not fit the bill?

TrevorDecker, 2016-04-01:
I think it could but I also don't want to swap controllers 14 days before a
demo.   But it would make for a great "mic dropping" aprill fools day joke