CMU-Robotics-Club/RoboBuggy #417: surface slows down after running alice for a few minutes

TrevorDecker, 2016-04-10:
We appear to have a memory leak which brings the entire surface to a slow speed

VasuAgrawal, 2016-04-11:
To reiterate some of points I made earlier when using Jprofiler on master, the majority of the memory leaks seemed to be in the form of char[], and to a lesser degree, int[]. It seemed that there was quite a bit of activity going on in the camera library, which might explain the int[], and commenting out the graphs definitely reduced memory footprint by a sizeable percentage. These probably aren't the only two sources of memory leaks, but it's a start. It also seemed like there was something in the server instance I had going, but that's since been purged from master in favor of a much simpler socket connection.

VivaanBahl, 2016-04-13:
I've narrowed this down to the map, it's very clear when we are adding points to the map, the system has a major performance drop. I will attempt to look into this, but I strongly recommend disabling the map for any future rolls until this issue has been properly fixed

VasuAgrawal, 2016-04-18:
It doesn't look like we had this issue on raceday.

VivaanBahl, 2016-04-18:
So there are a couple of problems that I have narrowed it down to, and I think it may be a combination of all of these that was contributing to the slowdown:

1. The localizer started at (0, 0), and the map tried to draw it, causing the map to try to render every single tile between here and Africa.
2. Drawing the camera's current image to the gui caused a major slowdown as well
3. There were a couple of branches that were trying to draw pose twice on the map, and was drawing both pose and GPS on the map.

Since we disabled the camera feed, and started the localizer to be on campus, this explains why we didn't have this issue on raceday

VivaanBahl, 2016-11-29:
I'm voting to close since we've taken the precautions described above, and are eventually moving away from ui-intensive operations on the Surface

closed 2016-11-29 by VivaanBahl